vaughnbetz/COFFE #18: Wrong simulation for connection block

MohamedEldafrawy, 2018-07-07:
https://github.com/vaughnbetz/COFFE/blob/7b407abe9fbbfa5ec5afa4b39aca344af62433ef/coffe/top_level.py#L129-L137

In the code snippet shown above, the top spice file that simulates the connection block is actually simulating the switch block. Normally the multiplexers in the connection block are larger than the multiplexers in the switch block. That's why this bug leads to underestimating the delay of the connection block. Unless there is a clear reasoning for this, it should be changes to cb_mux_on instead of sb_mux_on.

sadegh68, 2018-07-08:
The "sb_mux_on" in the code snippet is not the circuit being evaluated. It is part of the context for "routing_wire_load" (it includes a cb mux) which is being evaluated.

Therefore this is not a bug. Please file such issues under "help wanted" or "question".

closed 2018-07-08 by sadegh68

MohamedEldafrawy, 2018-07-08:
That's true thanks for the clarification.