vaughnbetz/COFFE #20: Bug at Finfet Case

basakozaydin, 2019-08-08:
Hello.
The basic_subcircuits.py script in the coffe subfolder has a small bug which prevents finfet option from running. In finfet cases of nand2_generate, nand2_generate_lp, nand3_generate and nand3_generate_lp functions within this script declares all of the finfets as in the below:
`spice_file.write("MNDOWN2 n_out n_vdd n_z n_gnd nmos L=gate_length nfin=Wn\n")
 spice_file.write("ASEO=Wn*min_tran_width*trans_diffusion_length ADEO=Wn*min_tran_width*trans_diffusion_length PSEO=Wn*min_tran_width+2*trans_diffusion_length PDEO=Wn*min_tran_width+2*trans_diffusion_length\n")`
Here, there shouldn't be a newline character after the "nfin=Wn" expression. With the presence of newline character HSPICE interprets ASEO as an element and yields the error "UNKNOWN ELEMENT ASEO".

vaughnbetz, 2019-08-08:
Thanks for the detailed report.  Ibrahim, can you push the patch?

Ibrahimai, 2019-10-18:
Removed the new line and the generated netlist can be simulated now.

closed 2019-10-18 by Ibrahimai