vaughnbetz/COFFE #27: Data port declarations in XML

Alimellat, 2020-07-09:
Sometimes the data port declarations inside the generated XML file are not properly closed. For instance,
```xml
 <port name="data" clock="clk">
```
is in the generated files and leads to an error when being parsed by an XML parser. It should probably be like:
```xml
 <port name="data" clock="clk"/>
```

vaughnbetz, 2020-07-09:
Thanks Alireza.  Does this lead to an architecture file snippet that won't go into vtr, or just into more strict xml parsers?  Good to fix either way, but a more critical issue if it won't go back into vtr.

Alimellat, 2020-07-09:
No problem. It does not go into vtr. This is the error I got from vtr:
 ```zsh
 Invalid tag sequence: <port></input_ports> at /mnt/test/vtr/vtr-verilog-to-routing/vtr_flow/scripts/run_vtr_flow.pl line 399
```

vaughnbetz, 2020-07-09:
I think the bug is on two lines:  986 of vpr.py and line 1562 which both look the same:

vpr_file.write("  <port name=\"data\" clock=\"clk\">  \n")

Should be:

vpr_file.write("  <port name=\"data\" clock=\"clk\"**/**>  \n")

I don't have hspice etc. set up to easily test a change -- can you make this change and try it out?  If it works (and I'm pretty sure it will, please file a pull request).

Alimellat, 2020-07-09:
Sure. I will test it and let you know about the results.

Alimellat, 2020-07-10:
It seems like the problem still persists. I spotted a couple of more lines in the vpr.py that had the same problem and modified them. I will try the new one out now and let you know about the result.

Alimellat, 2020-07-10:
The new version seems to be doing ok. I tested it with flut0.txt input file and the output was ok. I have submitted a pull request.

closed 2020-07-27 by Alimellat