vaughnbetz/COFFE #29: lut connection in the generated xml file

Alimellat, 2020-07-27:
Hello,

When I check the XML  file generated for VPR, there is no connection to the  lut.in[2:2] in the following code block:
```xml
  <interconnect>
    <direct name="direct0" input="ble6.in_A" output="lut6.in[0:0]">
      <delay_constant max="1.5215e-10" in_port="ble6.in_A" out_port="lut6.in[0:0]" />
    </direct>
    <direct name="direct1" input="ble6.in_B" output="lut6.in[1:1]">
      <delay_constant max="1.5292e-10" in_port="ble6.in_B" out_port="lut6.in[1:1]" />
    </direct><!-- comment by Alireza-->
    <direct name="direct3" input="ble6.in_D" output="lut6.in[3:3]">
      <delay_constant max="9.525e-11" in_port="ble6.in_D" out_port="lut6.in[3:3]" />
    </direct>
      <!--Clock -->
      <direct name="direct4" input="ble6.clk" output="ff.clk"/>
```
Is there any specific reason for that? I have added the "comment by Alireza" comment to the XML file.

Furthermore, I am wondering if it should be possible to use the generated XML with a simple Verilog design (like the blink module included in VTR package) for running VTR.

I am attaching the COFFE input, output, and log files.
input:
[flut08-4.txt](https://github.com/vaughnbetz/COFFE/files/4979861/flut08-4.txt)
log file:
[alireza8-4.log](https://github.com/vaughnbetz/COFFE/files/4979863/alireza8-4.log)
XML output (changed the format to .txt to be able to upload here.
[flut08-4-xml.txt](https://github.com/vaughnbetz/COFFE/files/4979862/flut08-4-xml.txt)

vaughnbetz, 2020-07-27:
Your architecture has asked for the LUTc input to be usable as a FF input, and for the FF to be able to drive LUTc. This is a common optimization in Intel/Altera FPGAs; I think the Lewis paper on Stratix in the FPGA Symposium describes it as "quick feedback mode" so you could check that paper, or a Stratix series datasheet, to see more about it.

This means there is a mux between the ble6.in_C signal and the LUT input, which is represented here:
        
      <mux name="mux1" input="ble6.in_C ff.Q" output="lut6.in[2:2]">
        <delay_constant max="9.516e-11" in_port="ble6.in_C" out_port="lut6.in[2:2]" />
        <delay_constant max="9.516e-11" in_port="ff.Q" out_port="lut6.in[2:2]" />  
      </mux>

Alimellat, 2020-07-28:
Interesting. I'll try to read the paper. Thanks!

closed 2020-07-28 by Alimellat