vaughnbetz/COFFE #36: About wordlinedriver area calculation in fpga.py

yc2367, 2022-10-17:
In coffe/fpga.py, line 3593 when calculating and updating the area for wordline drivers. 

At line 4529, the calculation only considers the areafac which equals to 2**self.row_decoder_bits. However, should this area be multiplied by an additional factor of 2 to account for the wordline drivers for 2 BRAM ports (i.e., 2 wordlines per BRAM row)? When instantiating the BRAM wordlinedriver for FPGA at line 5766, the area for wordlinedriver still doesn't account for this additional factor of 2 (unlike the RAM_decoder_area which is multiplied by 2 at line 5721).

vaughnbetz, 2022-10-18:
I'm not sure.  @sadegh68 @aman26kbm : any opinion?

sadegh68, 2022-10-18:
I think you're correct; it seems to be missing a factor of two.

sadegh68, 2022-10-18:
My apologies for closing the issue by accident. I have reopened it. Please simply multiply the word line driver area by 2.

closed 2022-10-18 by sadegh68

yc2367, 2022-10-18:
Thanks a lot for the clarification! I will read other BRAM parts and see if a factor of 2 is missing for other subcircuits. Will you update the code or should I create a pull request for this?

aman26kbm, 2022-10-18:
It'll be great if you can create a PR. Appreciate your help!