vaughnbetz/COFFE #43: About an error(HSPICE failed to run)When I run an example (flut0. yaml).

luck-codeer, 2023-02-13:
When I run an example (flut0. yaml), it has an error. I am thinking whether it is related to the working environment and version. I use ubuntu18.04.6, Hspice 2017.12 and COFFE2. So I want to ask if there are any environment and version restrictions on the use of COFFE2. I would be very grateful indeed for any help you could give me.
[image]

vaughnbetz, 2023-02-15:
@StephenMoreOSU : any ideas?

vaughnbetz, 2023-02-15:
One thing I have seen in the past is the license server becoming unavailable for a while, causing a run to fail.

StephenMoreOSU, 2023-02-15:
@luck-codeer I'm guessing its an issue with your hspice license, hspice is currently a dependency for running COFFE 2. If you can could you go to the "sb_mux" directory in the "arch_out_folder" (either specified in the configuration file or at <YOUR_COFFE_PATH>/COFFE/input_files/flut0/arch_out_dir) and show me the sb_mux.lis file? It should give some intutition on why the HSPICE run failed.

luck-codeer, 2023-02-15:
OK，This is sb_mux.lis file. According to what it said, I changed the path of line 38 in the process_data.l file, and then executed the code, the file path of process_data.l would be wrong again.Do you know where the source of the problem is?
[image]
[image]

StephenMoreOSU, 2023-02-15:
Yea so in the "flut0.yaml" input file there is a parameter which points to the spice models, in the unit test the default is for it to be set to ~/COFFE/spice_models/... so you need to set the "~/COFFE/" portion to the path of your coffe repo

StephenMoreOSU, 2023-02-15:
I'll leave this issue open until I update the docs to inform users where to change path specific parameters

luck-codeer, 2023-02-15:
Thank you for your help. The problem has been solved. It has troubled me for a week. Thank you very much!