vaughnbetz/COFFE #46: About the minimum supported temperature of COFFE2

luck-codeer, 2023-03-28:
Hi,
I am using your tool COFFE2 to do experiments on the performance (delay and power consumption) of FPGA components at low temperature. The results obtained are shown in the figure below. It can be seen that below -100 ℃, the delay will increase as the temperature decreases. 
[image]
However, according to this paper: "CryoWire: wire-driven microarchitecture designs for cryogenic computing." (https://dl.acm.org/doi/abs/10.1145/3503222.3507749), as the temperature decreases, the delay will also decrease, which is inconsistent with the experimental results.
So, I would like to ask you two questions. One is what is the minimum temperature supported by COFFE2, and the other is if I want to obtain a delay for components such as SB MUX at - 200 ℃. Do you have any suggestions.

vaughnbetz, 2023-03-29:
Threshold voltage increases as temperature drops while mobility increases. Hence there are two competing factors and the worst case performance can occur at low temperatures, particularly for pass gate logic in advanced technologies.Coffe has no lower temp limit so long as your spice decks are accurate at that T. Transmission gate logic and gate boosting will help at low T.Vaughn

luck-codeer, 2023-03-31:
Thanks for the reply, Prof. Betz. Your explanation has been very helpful for my question.

vaughnbetz, 2023-04-01:
Glad it helped!

Vaughn