vaughnbetz/COFFE #49: Question about whether COFFE supports energy_per_toggle in the architecture description file required for VTR

jdzhu19, 2023-05-20:
If wanting to do power estimation in VTR, it needs to have a power description with energy_per_toggle in the architecture description file, for example, energy_per_toggle of memory in various modes, I would like to ask if COFFE supports to calculate this value?

vaughnbetz, 2023-05-20:
I don't see an output line for that in the code, although I only did a
quick look. It should be possible to add if it isn't there though (although
you might have to do it).  Stephen & Sadegh: does COFFE include power data
in the vpr output file?

We should also add a regtest for vpr output; I don't see any golden test
results for a vpr file output which makes it harder to check the answer to
this question.

Vaughn

jdzhu19, 2023-05-21:
> I don't see an output line for that in the code, although I only did a quick look. It should be possible to add if it isn't there though (although you might have to do it). Stephen & Sadegh: does [...]

I was doing some custom coding on the vpr output file of COFFE some time ago, and I am sure that there is no information related to power in the original vpr output file, and there seems to be a lack of information like energy_per_toggle in the final report formed by COFFE, how should I go about getting this information?

vaughnbetz, 2023-05-21:
We compute energy per toggle by putting current monitors on the power
supply and toggling an input, integrating the current drawn over a rise + a
fall, and than taking Vdd * integral(curent) / 2 (since we performed two
toggles).  For memory it's a bit different; it's the energy per read or
write (it's tied to the read or write access, not how many inputs are
toggling).  White we've done analyses like that many times, I am not sure
how far into the output it is integrated, vs. what was done by hand in a
spice deck after transistor sizing.  Hopefully Stephen or Sadegh can
comment.

Vaughn

vaughnbetz, 2023-05-21:
Hello Vaughn,

Although we compute energy in fpga.py, we don't dump it in the architecture
file. Since that info is available in fpga.py, it shouldn't be too hard to
dump it.

Best,
Sadegh

vaughnbetz, 2023-05-21:
Great, thanks Sadegh.