vaughnbetz/COFFE #50: About the meaning of lut_input, the composition of a tile and how to measure the power consumption of a tile.

luck-codeer, 2023-05-26:
Hi, I have two questions to ask you,
First, in coffe/fpga.py, line 6409, I want to know why the delay of LUT equals the delay of lut_input plus the maximum delay between driver and not_driver. What refer to the delay of lut_input and driver in this context? Are there any relevant references on the composition of a tile in COFFE2?
[image]
Second, in report.txt, some modules have a power consumption value of n/a. I would like to know the specific reason why the power consumption of these modules has not been measured. If I want to get the power consumption of a tile, how should I measure it.
[image]

vaughnbetz, 2023-05-26:
I believe that the lut delay will be the path from the SRAM cell through the mux tree to the output.  The driver and not_driver are the buffers (inverter chains) that drive the pass tree forming the mux that is the heart of the LUT.  We need both the true and the complement of the input signal (e.g. LUT input a) so we have both a driver and a not_driver.

I'm not sure why the FF has an n/a power.  Perhaps its power was counted elsewhere (e.g. in the clock)?  @sadegh68 @StephenMoreOSU  any ideas?

StephenMoreOSU, 2024-04-16:
In this version of COFFE the flip flop is not actually sized and uses values found previously for 22nm. It doesn't have a testbench to measure its power and delay characteristics.