vaughnbetz/COFFE #53: About [R, C] in the metal layer

luck-codeer, 2023-07-25:
I would also like to ask you a question about how to obtain the resistance and capacitance values per unit length in this metal layer. If I want to obtain a value around -200 ℃, do you have any good suggestions. 
[image]

luck-codeer, 2023-07-25:
Ok, thank you Professor Betz, I checked the resistivity of copper at low temperature before, but I want to ask if the capacitance per unit length needs to be modified at different temperatures. Also, can you take a look at my last question(https://github.com/vaughnbetz/COFFE/issues/52), because Straix10's architecture is very important in COFFE and VTR.

luck-codeer, 2023-07-25:
okay, thank you, Professor Betz. I understand

StephenMoreOSU, 2024-04-16:
You should also set the temperature in HSPICE to make sure the active elements (transistors) are using the same temperature you expect to be simulating. The ".TEMP X" directive can be used for this. I'm not sure if setting the temperature in HSPICE also does some conversion to passive elements but [this thread](https://www.edaboard.com/threads/simulation-temperature-and-extraction-temperature.393536/) seems to have more information about it.