vaughnbetz/COFFE #55: About the issue of abnormal power consumption evaluation at low temperatures

luck-codeer, 2023-10-26:
Hello, Professor Betz. I apologize for bothering you again. 
I have encountered some phenomena in my work on designing FPGA architectures at low temperatures using COFFE and VTR. After consulting the VTR source code and corresponding papers, I still don't understand this phenomenon. I have also posted a related issue on VTR(https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/2424), but no one has responded to me yet, so I wanted to ask if you have any suggestions. 
Here is my question: At temperatures close to -200°C, I ran the same benchmark on the same FPGA chip (with the same design parameters, using fixed_layout, channel width of 96, vth0=0.3). The only difference between the two sets of experiments was that one had a power supply voltage of 0.5 and the other had 0.6. During the experiment, the power consumption decreased as VDD increased, and the dynamic power consumption in both sets of experiments was very small. Here is the report section of the two experiments.
[image]
[image]

This phenomenon is not consistent with the conclusion of the paper(https://dl.acm.org/doi/abs/10.1145/3307650.3322219). Theoretically, the static power consumption of an FPGA at low temperatures should be extremely low according to the following formula, which means that the dynamic power consumption ratio should be close to 1. 
[image]

But the experimental results are not like that. Also, with the increase in VDD, the power consumption should increase, but there was a sudden drop in this experiment. I have searched through the relevant source code and papers of VTR, but I haven't found a suitable explanation, so I wanted to ask if you have any suggestions.

luck-codeer, 2023-10-26:
It is worth noting that when I set vth0 to 0.2, the power consumption increases as VDD increases, although the dynamic power consumption ratio is still not close to 1, and even the ratio is very low. Below are the results of the relevant experiments I did at VDD = 0.2.
[image]

vaughnbetz, 2023-10-28:
That does seem strange. The static power is largely going to come from COFFE; VTR just adds up the numbers.  Are your spice models accurate for static power at -200 C?  I think you should run some SPICE runs by hand and validate (1) the SPICE models and (2) the COFFE output.

For the low dynamic power: what toggle rates are you getting and how did you compute them?  If you don't have reasonable toggle rates you won't get a reasonable dynamic power.

luck-codeer, 2023-10-30:
Hi, Professor Betz.
I understand the reason now. There was a problem with the leakage current of a line in the 22nm.xml file generated by generate_cmos_tech_data.pl, and I have fixed it. Thank you, Professor Betz.

vaughnbetz, 2023-10-31:
Thanks.  If this is a general problem others would face, it would be great if you can make a PR to land the fix.

luck-codeer, 2023-11-05:
hello, professor Betz. I have already submitted the relevant pull request, but I still have some questions on this issue(https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/2424). Could you provide me with some advice?