vaughnbetz/COFFE #58: HSPICE failed to run while updating delay for sb_mux

ketan170102030, 2024-04-16:
While running an example flut0.yaml, it is failing in delay updating stage although files are generating in the output directory but thing to note is sb_mux.lis file is empty.
I am using COFFE 2 tool on windows 10 and hspice 2023.03
Please help me to understand and solve the issue.
[image]
[image]

vaughnbetz, 2024-04-16:
@StephenMoreOSU might have an idea, or perhaps his more recent code updates will fix this. He is in the middle of a big COFFE upgrade/refactoring.

StephenMoreOSU, 2024-04-16:
Hi @ketan170102030,

As @vaughnbetz mentioned I'm in the midst of trying to refactor a lot of the COFFE codebase with accompanying documentation. ETA for this will be around a month.

For one I'm not entirely sure if COFFE is supported for windows, I've only ever run it from a Linux machine, I imagine that if there would be some errors related to OS on this version of COFFE it would be path related (some hardcoded path somewhere in code). But I'd recommend using a linux environment. 

The next thing is to make sure you have HSPICE on your system, you could try running the following:
`bash
hspice -v
`
And you should get something like the following:
[image]

If you have not installed HSPICE it will not work as it's a key dependency.

Sorry I've been absent from answering issues lately, will go through them today...

ketan170102030, 2024-05-30:
Hi @StephenMoreOSU I am trying to install HSPICE on ubuntu 22.04 but could not install it correctly. Can you point to any reference guide that I can refer to specifically for ubuntu ? Also may I ask which Linux environment you are using?

StephenMoreOSU, 2024-10-27:
HSPICE installation will have to be done through Synopsys provided documentation (protected by license agreement), you need a valid license and probably some version of [FlexLM ](https://en.wikipedia.org/wiki/FlexNet_Publisher) (depending on version) to be able to install and run HSPICE.

StephenMoreOSU, 2024-11-04:
Closing this as its been dead for a while, assuming the issue was a lack of HSPICE installation on system (which is out of bounds at the moment)

closed 2024-11-04 by StephenMoreOSU